ms-iot/BusProviders #5: Project does not compile

jamessdixon, 2015-09-09:
After pulling down a forked repo, I tried to compile the 2 CPP projects (PwmPCA9685 & PwmSoftware).  It says it successfully compiles, but no .dll is generated in the .bin folder.  Also, the references from the SamplePWMConsumer project are broken.
[image]

Recompiling does not fix it.  Trying to remove and reattach the references also throws an exception.

jessekaplan, 2015-10-12:
We've not been able to reproduce this problem. You should be able to find the binaries (for an ARM build) in this path: BusProviders-develop\PWM\ARM\Debug

Have you tried creating a new C++/CX project (for desktop)? It's possible this is a configuration issue. 

Thanks,
Jesse

ooeygui, 2015-12-07:
We cannot repro this issue

closed 2015-12-07 by ooeygui