ms-iot/BusProviders #13: Lightning Providers Arduino Dependency = Only Runs On Arduino?

CodeChief, 2015-10-19:
Do these drivers run on normal RasPi2 IoT Core machines without any Arduino? The way it's presented on the IoT download page (download link with big Arduino logo) and NuGet dependency suggests not.

msalehmsft, 2015-10-19:
The providers will run on Raspberry Pi2 or MBM with Windows IOT Core only. They won't run on an Arduino nor they're designed to communicate with one.
The main goal of the providers is to interface with a new Direct memory mapped driver that provide a huge boost in the performance of bus controllers, including GPIO, SPI and I2c on Raspberry Pi 2 or MBM.

Note, there are two other open-source libraries targeting Arduino devices:
* [Windows Remote Arduino](http://ms-iot.github.io/content/en-US/win10/SetupPCWRA.htm): an open-source Windows Runtime Component library which allows Makers to control an Arduino through a Bluetooth or USB connection
* [Windows Virtual Shields for Arduino](http://ms-iot.github.io/content/en-US/win10/SetupPCWVSA.htm): an open-source library primarily for the Arduino UNO which communicates with a universal app running on a Win10 device, such as a Lumia phone. The library exposes Lumia phones' sensors and capabilities to the an Arduino Wiring Sketch.

CodeChief, 2015-10-20:
First thanks for the quick and detailed response, but it is incorrect.

1. The documentation states the Ardino SDK is required.
2. The only public release, the NuGet 1.0.0-alpha package, *requires the Arduino SDK and only includes the AVR processor build*.
3. So it is *impossible to use the current release with IoT Core*, e.g. Raspberry Pi 2 (and probably not MBM either).

Please release a separate NuGet package with ARM support (for Raspberry Pi) and Intel support (for MBM) and no Arduino dependency. I'd also recommend putting the documentation/download links in the relevant RasPi2 and MBM sections (it is currently under Arduino/wiring components and states Arduino SDK is required).

msalehmsft, 2015-10-20:
Thanks for your suggestions!

Agree, the docs need more clarification in that regards. We are working on fixing it!

The dependency on the Arduino SDK, or more specifically on the [`Microsoft.IoT.SDKFromArduino`  NuGet package](https://github.com/ms-iot/arduino-sdk), is needed because the `Microsoft.IoT.Lightning` Nuget package references some of its declarations from `Arduino.h`. That enables support for the same structures and functions defined in the [Arduino Language Reference](https://www.arduino.cc/en/Reference/HomePage) (hence the Arduino logo), but for use on an IOT Core device rather than an Arduino.

In the future, we may able to refactor the code to eliminate the need of that dependency. 

It's also worth noting, the Arduino constructs/functions defined in `Microsoft.IoT.Lightning` library are used in a new VS project template to enable makers to create <i>Arduino Wiring</i> sketches that runs on Raspberry Pi2 or MBM. The VS extension with the project templates can be downloaded from [here](https://visualstudiogallery.msdn.microsoft.com/55b357e1-a533-43ad-82a5-a88ac4b01dec).

[image]

[image]

And, you are correct, the intent of the ` Microsoft.IoT.Lightning` library is to be used as source code included in higher level components, such as the library these comments belong to, `Microsoft.IoT.Lightning.Providers`. 

However, binary nuget packages for both, `Microsoft.IoT.Lightning.Providers` and `Microsoft.IoT.Lightning.Providersis` will certainly be much easier to include in a project. Again, thanks for the suggestion!

CodeChief, 2015-10-21:
OK, all clear now, thanks :-)

closed 2015-10-21 by CodeChief