ms-iot/BusProviders #18: Cannot create more than one SPI device from SPI controller on Raspberry PI 2

Norbert62, 2015-12-29:
The following code throws an unhandled System.IO.FileLoadException with message "The process cannot access the file because it is being used by another process." when using the lightning library.The catch block is not entered.

        private async Task TestInitSpiDevices()
        {
            try
            {
                SpiController controller = await SpiController.GetDefaultAsync();     /* Get the default SPI controller */

                var settings1 = new SpiConnectionSettings(0); /* Create first SPI device                               */
                settings1.ClockFrequency = 3000000;
                settings1.Mode = SpiMode.Mode0;    

                var spiDevice1 = controller.GetDevice(settings1);             /* Get the Spi Display device using its settings */

                var settings2 = new SpiConnectionSettings(1); /* Create second SPI device                               */
                settings2.ClockFrequency = 3000000;
                settings2.Mode = SpiMode.Mode0;

                var spiDevice2 = controller.GetDevice(settings2);

            }
            catch (Exception exc)
            {
                string message = exc.Message;
            }            
        }

Creating only one device works without any error. Platform is Raspberry Pi 2 Model B running Windows IoT 10.0.10586. The code runs flawlessly with the Inbox driver.

msalehmsft, 2016-02-19:
Thanks for finding this issue! 
The issues is being caused by the code initializing the controller device. It's currently set to exclusive, which results in the exception thrown when accessing the same controller device more than once. 
We will look into a fix for this ASAP.

msalehmsft, 2016-06-16:
Fixed in the latest Lightning SDK, v1.1.0-alpha

closed 2016-06-16 by msalehmsft