ms-iot/BusProviders #19: Reading SPI with TransferFullDuplex method does not deliver any data

Norbert62, 2015-12-29:
The following code delivers only empty bytes in resultBuffer. deviceCmd is a SPI device on line 0 using the lightning provider:

        public ushort ReadCmd(byte address)
        {
            byte[] readCmd = { 0x3, address, 0, 0 };
            byte[] resultBuffer = new byte[4];

            this.deviceCmd.TransferFullDuplex(readCmd, resultBuffer);
            ushort result = (ushort)(resultBuffer[3] + (resultBuffer[2] << 8));

            return result;
        }

The test device is a vlsi codec chip. The same method run with the Inbox driver delivers data as expected. Method TransferSequential does not work either with lighning provider.

msalehmsft, 2016-02-19:
Thanks for finding this issue! 
Can you please provide more context into the issue? 
As self-contained source code, as well as details about the device used would be very helpful.

Thanks!

Norbert62, 2016-02-20:
The bug was detected using a vlsi 1503 codec chip. Trying to read a value from a command register using full duplex SPI transfers failed with the lightning code. It works flawlessly with the builtin spi driver.

IoTGirl, 2017-03-20:
Closing after review by @msalehmsft - Unable to reproduce this issue.

closed 2017-03-20 by IoTGirl

chaholl, 2017-04-06:
I've experienced the same issue. The problem can be replicated by following this hackster project and enabling Lightning drivers: https://www.hackster.io/dotMorten/windowsiottouch-44af19