ms-iot/BusProviders #21: I2C Bus Speed too fast when set to 'Standard' mode, resulting in data corruption

NameOfTheDragon, 2016-01-05:
When using an I2C device with the bus speed set to 'Standard' mode, the bus speed used by the Lightning provider is too fast and exceeds the 100KHz maximum, resulting in data corruption.

I create my I2C settings like this:

        public static I2cConnectionSettings ConnectionSettings(int address = DefaultSlaveAddress)
        {
            return new I2cConnectionSettings(address) {BusSpeed = I2cBusSpeed.StandardMode};
        }

With the default 'inbox' driver, that produces bus activity like so (note the SCL frequency measures dead on 100KHz, which is the maximum allowed speed in Standard mode):

[image]

With the lightning provider enabled, the same capture looks like this (note, SCL measures about 165KHz, way too fast!)

[image]

I've tried the exact same code (driving an MLX90614 temperature sensor) with both drivers and with the lightning driver enabled, it produces meaningless data which fluctuates wildly. Looking at the protocol analysis, it is clear that things are just not working correctly.

NameOfTheDragon, 2016-01-05:
This of course is a SHOWSTOPPER because the I2C bus simply doesn't work, with no possible fix in software.

Adelaiglesia, 2016-02-13:
I can reproduce this

msalehmsft, 2016-02-19:
Thanks for finding this issue! 
Looks like this is caused by the wrong clock divider value used.
We will look into a fix for this ASAP.

msalehmsft, 2016-03-04:
The clock divider was not set correctly. In the Lightning package source '\packages\Microsoft.IoT.Lightning.1.0.2-alpha\build\native\include\BcmI2cController.h' , can you try changing the value of the const ULONG CDIV_100KHZ to 2500 instead of the current 1500, then rebuild the project?

NameOfTheDragon, 2016-03-12:
I see you have already published a new NuGet package, 1.0.3-alpha. I've updated my solution to that version and it's spot on 100KHz now, so I'd say that is a fix. Many thanks for that!
[image]

P.S. How about pushing the package for the Lightning Provider so I don't have to deal with the C++ source code for that...? There's a NuSoec in the solution but it's not on NuGet anywhere.

msalehmsft, 2016-03-14:
Thank you for confirming!

Apologies for the delay in releasing the Nuget providers. We've been planning to publish it on Nuget for some time now, but it's been delayed by some prerequisites for publishing binaries. Hopefully, it's all resolved now and we'll be able to release it shortly, this week or next week.

NameOfTheDragon, 2016-03-15:
That will be helpful – thanks!

However, did you notice that I found what appears to be another showstopper in the I2C provider? Basically, with the lightning driver, you can only create one instance of an I2cDevice. Since I2C is meant to be a multi-device bus, that’s a bit of a problem.

Please see https://github.com/ms-iot/BusProviders/issues/32

Best regards,
Tim Long

msalehmsft, 2016-03-15:
It's been a known issue and the fix has been pulled to the [Lightning repository](https://github.com/ms-iot/Lightning).
Unfortunately,  it didn't make it to the latest Lightning Nuget, 1.0.3-alpha.

However, the fix is straightforward to apply. Either, copy [DmapSupport.cpp](https://github.com/ms-iot/lightning/blob/develop/source/DmapSupport.cpp) from the Lightning repository to your Lightning source folder; normally <solution root>\packages\Microsoft.IoT.Lightning.1.0.3-alpha\build\native\source, or just update this [one line](https://github.com/ms-iot/lightning/commit/2a1e0dfb8858f47616798921ac18fbf09639e7b6) in the same file.

If thanks fixes the issue for you, please let me know.
Thanks

msalehmsft, 2016-06-16:
Fixed in the latest Lightning SDK, v1.1.0-alpha. If still not fixed, please reopen the issue.

closed 2016-06-16 by msalehmsft