ms-iot/BusProviders #24: Remove Un-necessary Internet Client Capability from Lightning Provider

CodeChief, 2016-01-13:
Please remove the "internet client" capability from the Lightning Provider before you deliver a final NuGet package with the consumer WINMD (I know currently the NuGet package just has C++/Arduino dependencies).

I presume you are delivering the Microsoft.Iot.LightningProvider.winmd to NuGet, else it's very cumbersome to have to build and copy interim code just to use the new provider.,

Anyway the point of concern is that users/customers will think our hardware apps want to access the internet. So following the security best practice of least privilege (also preached by MS) we must not claim something we do not require.

I'm talking about removing the "<Capability Name="internetClient" />" of course. At least from the lightning provider c++ project, but best from all samples to demonstrate the best practice.

msalehmsft, 2016-01-13:
Thanks for pointing out the "internetClient" issue. You're right, It' not needed, and we'll remove it.
Yes, we are planning to deliver a Nuget containing the WinRT component. You may have noticed, the script and  to create the Nuget has already been committed under BusProviders/Microsoft.IoT.Lightning.Providers/Nuget, so the next step is to publish it.

msalehmsft, 2016-03-14:
This has been fixed.

msalehmsft, 2016-05-24:
The latest sample code updates have these capabilities removed.

closed 2016-05-24 by msalehmsft