ms-iot/BusProviders #25: IoT Capabilities Not Visible in Visual Studio Package Manifest Editor

CodeChief, 2016-01-13:
Following on from #23 can we get these "hidden" capabilities (the "lowLevelDevices" and the device GUID) into the VS editor? I appreciate if this is not for this repository then please tell me where it should be logged. Last time I tried to raise an issue about an MS SDK with MS Connect for Visual Studio they just closed it as "external" without any explanation or come-back.

Failing an editor, perhaps you could inject a build warning (actually it's really an error if you use the Lightning provider) as a workaround to inform the developer (and save a lot of wasted time and hair pulling).

bfjelds, 2016-01-13:
Thanks for the feedback!  We are definitely aware of this omission in Visual Studio's UI.  We are working with Visual Studio to fit this into their schedule.

IoTGirl, 2017-06-28:
Closing Stale - Issue was raised with VS.

closed 2017-06-28 by IoTGirl