ms-iot/BusProviders #29: Speech synthesis in universal app doesn't work when lightning providers used.

delamitry, 2016-02-19:
I'm not sure where to log this issue but the speech sythesizer doesn't seem to work when I enable the Direct Memory Mapped Driver on the rpi2.  When I change back to Inbox Driver, starts working again.

delamitry, 2016-02-19:
This is the simplest example I could come up with that works fine before enabling the lightning providers.
`var synthesizer = new SpeechSynthesizer();
var stream = await synthesizer.SynthesizeTextToStreamAsync("Hello World");
BackgroundMediaPlayer.Current.SetStreamSource(stream);
BackgroundMediaPlayer.Current.Play();`

delamitry, 2016-02-28:
I've determined that it is only affecting the headphone jack on the rpi2.  HDMI audio is working along with a usb audio.

jordanrh1, 2016-02-29:
The issue is that dmap.sys is being installed on the PWM device node, which is used for audio. 

To reinstall the correct driver on the PWM device node, run

reg add HKEY_LOCAL_MACHINE\SYSTEM\DriverDatabase\DeviceIds\ACPI\BCM2844 /v dmap.inf /t REG_BINARY /d 02ff0100
reg add HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\enum\ACPI\bcm2844\0 /v ConfigFlags /t REG_DWORD /d 0x20
devcon restart acpi\bcm2844

Verify that correct driver is running on PWM node:

C:\Data>devcon status acpi\bcm2844
ACPI\BCM2844\0
    Name: BCM2836 PWM Controller
    Driver is running.
1 matching device(s) found.

A fix for the issue will be available in the next insider preview. Thank you for reporting it.

IoTGirl, 2017-01-30:
Over 6 months with no new input - Closing

closed 2017-01-30 by IoTGirl