ms-iot/BusProviders #32: Lightning provider allows only one instance of I2cDevice

NameOfTheDragon, 2016-03-13:
When using the lightning provider, as soon as I try to create a second instance of I2cDevice, I get the following error:

> The program '[0xA34] backgroundTaskHost.exe' has exited with code -1073741189 (0xc000027b).

This is at variance with the inbox provider, which works as expected.

Unfortunately this means that the lightning provider prevents communication with more than one I2C device, unless all other instances are disposed and re-created every time they are used. This is no good at all!

The problem can be reproduced using the following complete UWP Background Application, which works with the inbox provider but fails with the lightning provider:

using System;
using System.Threading.Tasks;
using Windows.ApplicationModel.Background;
using Windows.Devices;
using Windows.Devices.I2c;
using Microsoft.IoT.Lightning.Providers;

// The Background Application template is documented at http://go.microsoft.com/fwlink/?LinkID=533884&clcid=0x409

namespace I2cConcurrencyTest
    {
    public sealed class StartupTask : IBackgroundTask
        {
        public void Run(IBackgroundTaskInstance taskInstance)
            {
            if (LightningProvider.IsLightningEnabled)
                {
                // Set Lightning as the default provider
                LowLevelDevicesController.DefaultProvider = LightningProvider.GetAggregateProvider();
                }

            var controller = I2cController.GetDefaultAsync().AsTask().WaitFoResult();
            var device1Settings = new I2cConnectionSettings(0x50);
            var device1 = controller.GetDevice(device1Settings);
            var device2Settings = new I2cConnectionSettings(0x60);
            var device2 = controller.GetDevice(device2Settings);
            }
        }

    public static class TaskExtensions
        {
        internal static TResult WaitFoResult<TResult>(this Task<TResult> asyncOperation)
            {
            asyncOperation.Wait();
            return asyncOperation.Result;
            }
        }
    }

msalehmsft, 2016-03-15:
Adding my comment from [#21](https://github.com/ms-iot/BusProviders/issues/21) here as well.

The fix has been pulled to the [Lightning repository](https://github.com/ms-iot/Lightning).
Unfortunately,  it didn't make it to the latest Lightning Nuget, 1.0.3-alpha.

However, the fix is straightforward to apply. Either, copy [DmapSupport.cpp](https://github.com/ms-iot/lightning/blob/develop/source/DmapSupport.cpp) from the Lightning repository to your Lightning source folder; normally <solution root>\packages\Microsoft.IoT.Lightning.1.0.3-alpha\build\native\source, or just update this [one line](https://github.com/ms-iot/lightning/commit/2a1e0dfb8858f47616798921ac18fbf09639e7b6) in the same file.

If thanks fixes the issue for you, please let me know.
Thanks

NameOfTheDragon, 2016-03-15:
I can give that a go. 

How does that line of code relate to the sharing mode specified in the I2cConnectionSettings? It isn't clear to me whether the driver is using shared mode for the device, or the controller. The abstraction is that there is an I2C Controller, which can create I2cDevices; each I2cDevice corresponds to an endpoint/slave device on the bus and the controller corresponds to the bus itself. Thus, it should be possible to create multiple instances of I2cDevice each with exclusive access, provided each has a unique slave address, should it not?

I suppose the question is whether exclusivity is at the controller or device level.

msalehmsft, 2016-03-15:
With the current implementation of the I2C Lightning provider, every time a new I2CDevice is created, the code creates a new device driver object (CustomDevice::FromIdAsync) to send/receive the IOCTLs needed to send the I2c device transactions. When the share mode was set to exclusive, only the 1st call succeed and subsequent ones failed.

Ideally, we should make that call at the I2cController level instead. And, reuse that CustomDevice object for all I2CDevice's, but it's not the case yet. We're working on refining that.

The slave address itself is only set at the transaction level.

So, to answer your question, currently the exclusivity (to open the CustomDevice device driver) is at the I2CDevice level.

Note, we still had to set the CustomDevice mode to 'Shared', because there are solutions that open connection to devices from multiple projects (DLLs). So, even if we move the exclusivity to the I2cContoller, those calls would still fail, as each project would still create its own I2CController object, unless the CustomDevice is shared of course.

NameOfTheDragon, 2016-03-15:
Thanks for the clarification. I think this is OK, if I understand correctly (I don't pretend to understand the complexities of Windows device drivers). My expectation as a developer is that in Shared mode, I want to be able to have multiple threads all talking to a slave device, but in exclusive mode only one at a time. I think that's what you've said...

msalehmsft, 2016-03-15:
Just to clarify, the Lightning providers library does NOT honor the I2cConnectionSettings.SharingMode correctly today. It'll treat all I2cConnections as 'shared'. This is something we're working on fixing as well.

AlfoldiMate, 2016-05-04:
Is there any solution yet about how I could communicate with two different I2C device?

msalehmsft, 2016-05-04:
Hi AlfoldiMate, Can you try upgrading to the latest Lightning SDK, 1.0.4 (uploaded last week), from your Visual Studio project? It enables driver sharing, so should resolve this issue.

Related to that, we've also updated the [WinRT Lightning providers](https://github.com/ms-iot/BusProviders/tree/develop/Microsoft.IoT.Lightning.Providers/Providers) project with several bug fixes and enabled GPIO callbacks, so if you need it, you may want to grab that one too.

AlfoldiMate, 2016-05-05:
I updated to the latest Lightning SDK, but my Lightning provider was the latest. It was working for me, with one i2c device without the sdk. 
Here is my relevant code:
  ```
  public static class DeviceHandler 
    {
        private static I2cController controller;
        private static I2cDevice ADC;
        private static I2cDevice ACCL;

        private static List<Int16> datalist = new List<Int16>(new Int16[3]);
        public static List<Int16> DataList { get { return datalist; } }

        public async static Task Init()
        {
            controller = await I2cController.GetDefaultAsync();

            ACCL = controller.GetDevice(new I2cConnectionSettings(0x68));
            ACCL.ConnectionSettings.SharingMode = I2cSharingMode.Shared;
            ACCL.Write(new byte[] { 107, 0 });

            ADC = controller.GetDevice(new I2cConnectionSettings(0x48));
            ADC.ConnectionSettings.SharingMode = I2cSharingMode.Shared;
            ADC.Write(new byte[] { 0x01, 0x04, 0x00 });
        }
...
```
If I initialize just one I2cDevice it works fine, but I get blue screen, when I want to use both. Driver have been changed in the Device Portal, and the default provider have been set to the Ligthning provider. I do not know what can be the problem, but to continuously disposing the initialized device to use another one is too slow for my project.
GPIO also working with the Lightning provider for me. I would like to find a solution for the two I2cDevice problem.

msalehmsft, 2016-05-09:
Let me try the same. Blue screen could mean an issue with the driver. 
Can you tell me the version of IOT Core you're using?

msalehmsft, 2016-05-24:
Tried it with the Raspberry Pi Sense Hat and works fine when initializing two i2c devices (). I've used the Lightning binary nuget (Microsoft.IoT.Ligthning v1.1.0-alpha). Here's a fork of the RPI.SenseHat project that supports Lightning: [https://github.com/msalehmsft/RPi.SenseHat/commit/bcae785b921314eb8617b15e86c0f2aa62ee064a](https://github.com/msalehmsft/RPi.SenseHat/commit/bcae785b921314eb8617b15e86c0f2aa62ee064a). Please check LSM9DS1ImuSensor.cs for an example.

NameOfTheDragon, 2016-05-24:
OK, this package was published after I originally reported the issue, so I will give it a try and see if I can reproduce your success.

I can only see version 1.0.0 listed on the NuGet web site though… perhaps it doesn’t show pre-release versions. I’ll try it from Visual Studio ASAP and report back.

Thanks,
Tim

msalehmsft, 2016-05-24:
Thanks!
To view the pre release packages, please check the "Include prerelease" option in the Nuget package manager.

[image]

Also, after you install this Nuget, you will need to remove the Providers project from your solution, since this package replaces it.

NameOfTheDragon, 2016-06-02:
I'm not able to verify this because the last two Insider builds are very unstable on my Pi3. The penultimate build is unable to shut down or restart correctly. The latest build works on first boot but fails on subsequent boots until the SD card is reimaged. Any idea how I report this?

msalehmsft, 2016-06-02:
Can you try another SD card? May I suggest one of these: [this one](http://www.amazon.com/gp/product/B00IVPU786) or [this one](http://www.amazon.com/SanDisk-Ultra-Micro-SDHC-16GB/dp/9966573445).
If that doesn't work, can you please post this issue on the [Windows IOT forum](https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT)

NameOfTheDragon, 2016-06-02:
@Mahmoud I bought it new with the raspberry pi, there shouldn’t be anything wrong with it. It’s the SanDisk one that I have (your second link). And it works fine with the original IoT build. I have another one somewhere, I’ll give that a go and see if it is any different, just in case; but I really don’t think the SD card is the problem…
--Tim

msalehmsft, 2016-06-02:
@NameOfTheDragon Agree, if the SD card is working with another IOT build, then most probably it's a different issue. Can you try posting the issue to the [Windows IOT forum](https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT) ?

NameOfTheDragon, 2016-06-02:
Thanks. I tried the alternative SD card, a 32 Gb SanDisk Ultra HC-I and that’s made no difference. I still can’t shutdown/restart cleanly – I have posted on the forum and will see what we can discover there: https://social.msdn.microsoft.com/Forums/en-US/d0fa9ace-3458-456d-bc42-e01818d09ffa/insider-builds-untable-on-raspberry-pi-3?forum=WindowsIoT

Thanks,
Tim Long

IoTGirl, 2017-01-25:
Closing as stale and Microsoft was unable to reproduce the issue. Please reopen with full details of the issue if it still occurs.  For guidance we ask for the following starting details:

1. Full details of the hardware and software involved
2. Reproduction steps someone would take to see the issue
3. Actual Result with error codes etc
4. Expected result

closed 2017-01-25 by IoTGirl