ms-iot/BusProviders #33: Project status

aL3891, 2016-04-06:
Hello,
What is the current status of this project? it looks very promising but looking at the history there doesn't seem to be a lot of activity and its very cumbersome to use since its not published to nuget and the instructions are somewhat lacking.

jessekaplan, 2016-04-06:
Thanks for your interest here. We're working on improving the usability by adding Nuget packages and the instructions will be updated in sync. 

We recently added the "simulated hardware" provider to allow for app development/debugging when using the actual hardware is inconvenient. Beyond the improved usability of what we have, were there other improvements/additions you were hoping for? 

Thanks,
Jesse

aL3891, 2016-04-06:
I think the prime points of improvement would be nuget publishing and docs, its such a shame that you're doing all this work and its so hidden away. Being able to do closed loop and high perf timing (via interrupts) is super useful.

Its also a little confusing that this library has a dependency to arduino, you mentioned in another issue that you were thinking about factoring that out and i agree with that

msalehmsft, 2016-05-24:
We've completed another round of refactoring.
The code for the Lightning providers has moved to the [Lightning repository](https://github.com/ms-iot/lightning), while the binaries (for all Lightning SDK components) are now included and published as part of the latest [Lightning SDK Nuget package](https://www.nuget.org/packages/Microsoft.IoT.Lightning/1.1.0-alpha). 
So, apps that need Lightning only need to reference that Nuget package (plus update their AppX manifest). We're working on updating our docs as well.

closed 2016-06-16 by msalehmsft