ms-iot/BusProviders #34: abs macro in arduino.h violates compiling in release configuration in vs2015

siochs, 2016-04-21:
Hi, first I would like to refer to the following location at [https://social.msdn.microsoft.com/Forums/en-US/7537e5fb-a815-45df-ac1f-d12a03c91934/compiling-microsoftiotlightningproviders-in-release-configuration?forum=WindowsIoT](msdn forum).
The abs macro in Arduino.h prevents chrono std library to compile when switching from Debug to Release configuration in Visual Studio 2015. Deleting this macro (#define abs(x) ((x)>0?(x):-(x))) in line 123 abolishes this issue. Thanks

msalehmsft, 2016-06-16:
@siochs Thanks for finding this. It'll be fixed in the next version of the SDK

msalehmsft, 2016-09-28:
This has been fixed in the latest Lightning Nuget 1.1.0

closed 2016-09-28 by msalehmsft