ms-iot/BusProviders #39: Any ways to control ACK and NACK on I2C

AlexandrSurkov, 2016-07-27:
Hello! 
Is there any way to control ACK\NACK responses for I2C from C#?
I need to simulate some device that respond NACK for START condition when device is busy...

jordanrh1, 2017-01-30:
If the device fails to acknowledge its slave address, an exception will be thrown. If you wish to handle this in an exception-free way, use `I2cDevice.WritePartial/ReadPartial/WriteReadPartial`. They will return I2cTransferStatus.SlaveAddressNotAcknowledged if the slave address is not acknowledged.

It is not possible to simulate an I2C device on Windows because we support host mode only. We do not support I2C slave mode.

IoTGirl, 2017-01-30:
Answer given - Closing issue

closed 2017-01-30 by IoTGirl