ms-iot/BusProviders #44: Pi 3 - DmapSupport GetControllerBaseAddress for BCM2844 does not find device???

jamoeder, 2017-01-19:
I've been trying to enable either PWM0 or PWM1 using the proper alternate function calls to the relevant register on the BCM2835.  I think I have the pin configured properly but now I need to enable the PWM in the proper register in the BCM2844 chip.  I can see the "#define pi2PwmDeviceName" in the DmapSupport.h file but when I try to use the "GetControllerBaseAddress" function using the PWM device name it comes back blank.  I've looked in the dmap.inf file on the Pi and I see "%dmap.DeviceDesc%=dmap_Device,*MSFT8001,ACPI\BCM2841,ACPI\BCM2845,ACPI\BCM2838,ACPI\BCM2839" but there's no BCM2844?  Was this done on purpose?  Can anyone supply any help so I can write to that chips register to get any PWN working?

The software driven PWM isn't dependable enough for my setup...

Help Please (and thanks)

Jim

IoTGirl, 2017-01-25:
Hi jamoeder,

I reached out to a member of the IoT Core Team and have the following answer for you:

> Is there any reason he isn’t using the lightning driver rather than trying to do this directly to the HW?

He says SW PWM isn’t reliable enough, and I agree, but Lightning is HW PWM.

Please try the Lightning solution as it should meet your needs.
Sincerely,
IoTGirl

closed 2017-01-25 by IoTGirl

jamoeder, 2017-01-25:
Thanks so much for the reply.  I really appreciate it.

The HW PWM in the Lightning solution is for a breakout board (PCA9685).  Although this is not the preferred solution it will work but I have to buy new hardware which is something I'd like to avoid if possible.  Did they run out of time before the current IOT release or was there a specific limitation that stopped them from enabling the BCM2844 registers?  Is there any chance you can post the DMAP driver so it can be modified to add support for the chip?

Again,  Thanks for the response thus far.

Jim

IoTGirl, 2017-01-26:
Hi Jim,

I have the following responses for you 

> So it’s mostly due to extra HW then… Jesse provides background here [https://github.com/ms-iot/lightning/issues/35#issuecomment-218862919](https://github.com/ms-iot/lightning/issues/35#issuecomment-218862919) but basically it is because of the underlying design of the Pi PWM controller.
While I would still agree with Jesse, if he wants to look through the damp code it is located here https://github.com/ms-iot/lightning

jamoeder, 2017-01-27:
Great Answer!  Thank you (and Jesse) very much for the clarification.

So...  For fun and profit, what are the chances you can post the DMAP drivers so I can enable BCM2844 chip and allow access to the PWM registers?  If I can get it to work I'd be happy to share?