ms-iot/BusProviders #50: Support ValueChanged in Lightning Provider for MinnowBoard with Windows IoT build 16299

mrajaraghavan, 2018-02-07:
With reference to the case #26.

ValueChanged is not working for me. I'm using Minnowboard Turbot dual core with windows iot build 16299. I have added Microsoft.IoT.Lightning v1.1.0. I wanted to capture an interrupt continuously (which is generated at an interval rate of less than 20 micro seconds and operates at a Frequency of 50-60 kHz ) using a gpio as input pin. I'm able to set a gpio as output also. Now i need to use ValueChanged. Can anyone help me in this regard.

mrajaraghavan, 2018-03-07:
Awaiting any response from your end which will help me a lot in my work.
Thanks in advance.

saraclay, 2018-11-09:
Hello,

If this is still an issue for you, could you please repost this in our MSDN forums? We have staff there that are monitoring the forums and will be able to help you.

MSDN forums: https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

saraclay, 2018-12-14:
Please post this issue on MSDN forums:
https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

closed 2018-12-14 by saraclay