ms-iot/BusProviders #51: Network Adapter Priorities in IoT broken after creators fall update

Vincation, 2018-03-27:
Hi,

After posting on the MSDN forum I was redirected here.

We have a board with 2 LAN ports, WiFi and an MBIM modem running Windows IoT core.

After the recent update of Windows IoT core our combination of an ethernet connection on a LAN port and the MBIM modem stopped working.

Everything was working fine a few months ago. A recent Windows update (1709) however made sure that if you have an active ethernet connection without internet, Windows does not route the internet through the MBIM modem anymore.
After disconnecting the ethernet, internet through the MBIM modem works fine. Plug it back in and internet through the MBIM modem stops again. A ping test with forced routing works fine, but any other traffic is not routed through the modem. It seems to be an MBIM driver specific problem since the combination WiFi and LAN without internet works fine.

We tried several things like: leaving the default gateway blank, changing metric interface and managing network adapter priorities (which I understand is created especially for these situations and should work/has worked)

The problem is more thoroughly described here, but this thread seemed to have died quickly: https://social.technet.microsoft.com/Forums/windows/en-US/e20c2af2-f2c1-419c-a2a3-cd2d07d8e699/network-problem-with-mbim-after-upgrading-windows-10-1709?forum=win10itpronetworking#e20c2af2-f2c1-419c-a2a3-cd2d07d8e699

Info on the prioritizing of the connection (what now is broken) here: https://www.ghacks.net/2016/12/02/change-network-adapter-priorities-in-windows-10/

Is this a known issue? Is there a temporary work-around?

My first choice would be for Microsoft to fix this of course, since it's clearly a freshly introduced bug. 

To be clear: since we did not find a workaround this is a mayor blocking issue. 
It is preventing the roll-out of a pretty important field test of Windows IoT core consisting of approximately 100 devices and has now already proven to be a disaster for our final production timeline.

Any help is greatly appreciated!

Greetings

Vincation, 2018-03-27:
Only one LAN Port (OR wifi) without internet and a MBIM modem are enough to reproduce te problem. also on a normal Win 10 1709.

You put a static address on the lan cable. no gateway filled in. So "route print"/"netstat -rn" looks fine.

If the cable is connected, no internet trough the cellular modem.

Without a cellular modem: for example one LAN without internet. wifi with internet -> works fine. It gets broken in windows 1709 if a cellular (mbim) modem is in the game. Tested with a telit 910 v2 & Vodafone K5150 with the microsoft stock driver

saraclay, 2018-11-09:
Hello,

If this is still an issue for you, could you please repost this in our MSDN forums? We have staff there that are monitoring the forums and will be able to help you.

MSDN forums: https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

Vincation, 2018-11-12:
Hey Sara,

Thanks for following up on this. I forgot to close this one.
You got me an answer on the Windows IoT core docs Git in May which is still working.
https://github.com/MicrosoftDocs/windows-iotcore-docs/issues/343#issuecomment-388127051

closed 2018-11-12 by Vincation